kirek007/ws-osd-py #9: Directory already exists, error in red text

EnglishTurbines2, 2023-01-07:
Having created the OSD overlay files, and then clicking on render video, Error message in red text "Directory already exists, remove directory"? No idea what that means as the path is auto generating?
Hangs and fails to render video and overlay?
It's creating the OSD files...lots of them!
Frustrating..

kirek007, 2023-01-07:
Hey,
There is already a step by step video how to use it. Try to follow steos in video and tell me of it works for you.

https://youtu.be/we3F4rIXTqU

I know that red message is not accurate today. But this is a 2 step process: first you have to generate PNG frames, then click render video to get video with osd.

EnglishTurbines2, 2023-01-08:
Hi there,

                    Thanks for the link to the video...Looks like I'm 
doing it right after all. The error message in red made me think it 
wasn't going to work!!...Looks like I have to wait a while to process a 
10 minute flight. My (older) Win 11 Laptop is flat out doing that 
rendering though even with SSD...

Great job on the Application...I'll try it again now I know it's normal 
behaviour.

Thanks for your help!!!

Dave..😀

Sent via BT Email App

text (Issue #9)

https://youtu.be/we3F4rIXTqU <https://youtu.be/we3F4rIXTqU>

EnglishTurbines2, 2023-01-08:
Hi Marek,
                  Great, that video clip was useful. Turned out I was doing it right, but the red text error message and the apparent non active Rendering box led me to believe it wasn’t working!!
  Worked great...
                                May I suggest:
1) Make the Configurator Box a bit smaller, its a tight fit on my Laptop screen and dragging and dropping is a bit tricky to do.
2) Text on the file path strings are rather large.
3) Try and make the rendering active box look like its doing something.
4) Fix the red error message somehow...

                  Great effort, this tool is going to be so useful, I need to start using 30 fps though because my old HP Laptop isn’t very powerful and the rendering takes ages...lol.

 Coffees bought!!!
                                                                 Cheers....Dave...:)

From: Marek Ruszczyk
Sent: 07 January 2023 23:28
To: kirek007/ws-osd-py
Cc: EnglishTurbines2; Author
Subject: Re: [kirek007/ws-osd-py] Directory already exists, error in red text(Issue #9)

Hey,
There is already a step by step video how to use it. Try to follow steos in video and tell me of it works for you.
https://youtu.be/we3F4rIXTqU
I know that red message is not accurate today. But this is a 2 step process: first you have to generate PNG frames, then click render video to get video with osd.