kirek007/ws-osd-py #15: [Reqeuest] Naming the files different!

Pairan, 2023-01-20:
Currently the generated png files are named `ws_000000001.png` and so one for every file.

If you however fly for longer and one flight has several video files then this causes extra work to get together again.

Could the files be named in a pattern that includes the source file? e.g. `AvatarG0001_xxxxxx.png`

That would make it easier to attached the generated files to each other ( like if AvatarG0001+AvatarG0002+AvatarG0003 = ONE FLIGHT)
WIth the current version I would have to rename all files to get them in the right sequence  :)

kirek007, 2023-01-20:
Why not import all sequences and merge them in the video editor? (This is how I'm doing it)

Pairan, 2023-01-20:
Because the flight consists of more videos and as such would overwrite the first parts as they have Thronsäle names

Pairan, 2023-01-20:
Yes … I’m aware of that

kirek007, 2023-01-20:
> Yes … I’m aware of that 

That's why I don't see any additional work that you have to do because of that. I'll consider this improvement in next versions, but I don't see personally any value in it.