kirek007/ws-osd-py #16: hide sensitive values not working

iamwinq, 2023-01-20:
feature to hide sensitive info like gps, altitude and distance does not seem to work in my testing, ardupilot setup by the way..
[image]

kirek007, 2023-01-20:
Can you provide OSD file? Looks like Ardupilot is using a different icon id, that's why it doesn't work

keithluneau, 2023-01-20:
Here's a short video, osd, and srt files from the end of my last flight. It's Ardupilot plane firmware. Hopefully this helps! https://drive.google.com/file/d/1s8FFXXZor7RdorVWQqW20dhkVUXKITuI/view?usp=share_link

iamwinq, 2023-01-21:
Hey buddy! Thanks for this. Let me check it out.

Best regards,

Erwin

iamwinq, 2023-01-21:
AvatarG0008.mp4
<https://drive.google.com/file/d/1ChII2oOL04v3IkpVSASkNkbjfhDhUkpR/view?usp=drive_web>
here's the files by the way.

iamwinq, 2023-01-21:
by the way, i tried to  render Keith's file as well, also didnt hide the
distance and altitude as well

 [image: image.png]

iamwinq, 2023-01-21:
keith's video rendered, with hide sensitive data enabled, yet still shows
altitude and distance...

[image: image.png]

iamwinq, 2023-01-21:
here are the files by the way
https://drive.google.com/drive/folders/1XTTvcsqFXdGlyLp_f1zKYgFBAXjm48BG?usp=sharing

iamwinq, 2023-01-21:
i also tried rendering keith's video, with Hide sensitive data enabled, yet it still shows alt and distance
[image]

kirek007, 2023-01-21:
Hey guys, looks like hiding data works only for INAV and only for GPS and
home distance. I'll try to make it works for ARDU too in the next version
;)

iamwinq, 2023-01-22:
> Hey guys, looks like hiding data works only for INAV and only for GPS and home distance. I'll try to make it works for ARDU too in the next version ;) W dniu sob., 21.01.2023 o 02:46 iamwinq ***@***.***> napisał(a):
> […](#)

thanks again! Looking forward to it ;)

endymuhardin, 2023-02-10:
It's not working with Betaflight either. The only thing it hides is the throttle value :D 
GPS Coordinate and Altitude is still showing.

kirek007, 2023-02-10:
Right, I should write somewhere that only inav is supported 😁

WereWolfPL, 2023-02-10:
I don't know how osd data are structured and if it's possible, but it would be nice (in the future) to be able to select which data yout want to hide (or what data you want to display) in the generated files. Maybe it's an overkill but why not :)

avsaase, 2023-02-10:
It would be nice if the user can supply the grid cells that they want to mask. That way the user can determine from themselves what they consider sensitive and you don't have to rely on detecting specific OSD element in the .osd files.