kirek007/ws-osd-py #23: empty generated folder

grc1000, 2023-01-21:
I do the first step, render the osd and then I give ok to the message, the RENDER VIDEO WITH OSD tab was supposed to be visible, but it is deleted and the generated folder is empty. I've tried several times and nothing. what am I doing wrong? AvatarG0000.mp4 does not exist in the folder.
[image]

[image]

grc1000, 2023-01-21:
[image]

kirek007, 2023-01-21:
This is something related to the diacritic chars in path. Can you try to move your files to the different directory like 'C:\walksnail\' and try again?

grc1000, 2023-01-22:
OK. now it worked. I did the following... I only used the video files that I was going to render at the moment, I put them in a root folder on the windows desktop and I clicked on render and then it did the whole process by itself, that was the way it worked for me. min. the font file doesn't matter which folder it came from, because it didn't get in the way. another thing...the hardware acceleration even if experimental works very well. did it in 1/4 of the time if you didn't use the resource, approved. I think you can close the topic now. thanks for the app and for the help.
[image]
[image]
[image]
[image]

kirek007, 2023-01-30:
Related to https://github.com/opencv/opencv/pull/21961