kirek007/ws-osd-py #26: Zoom and offset sliders have no effect on the output

WereWolfPL, 2023-01-21:
In the latest beta rc version - the sliders have no effect on generated images and video.
The preview window works fine, but in the output files osd is in the default position.

kirek007, 2023-01-30:
Can you provide any log from console?

WereWolfPL, 2023-01-31:
> Can you provide any log from console?

It doesn't give any error, just standard info, like: 
```
DEBUG:root:Preview update requested.
DEBUG:root:Slider updated.
DEBUG:root:Preview update requested.
DEBUG:root:Slider updated.
DEBUG:root:Preview update requested.
DEBUG:root:Current: 200/8719 (fps: 315)
DEBUG:root:Current: 400/8719 (fps: 294)
```

The latest version (0.8 beta) didn't fix it for me. No matter what I set on the sliders, the generated osd looks like all the settings were set to 0.

kirek007, 2023-01-31:
Then I need as much details as you can provide ;)
What operating system ?
Is it happening for all video files?
Screen of application
Whole log from console
If this is happening only for some files then I need them.

WereWolfPL, 2023-01-31:
Ok, so...
System: windows 10 64-bit
It's happening for all of my files. I'm not sure if it matters, but I use walksnail vrx module with full size vtx. Tested with firmware 29.33.16 and 31.36.8(beta). It's definitely something with generating .png files, rendering of the video doesn't matter here.

Here are some screenshots - app's window (I tested it with different settings), one of generated frames and some screenshot of the video, full console log and two of my sample .osd files.

Let me know if I can send any more details.

[image]
[image]
[image]
[log.txt](https://github.com/kirek007/ws-osd-py/files/10545548/log.txt)
[osd_files.zip](https://github.com/kirek007/ws-osd-py/files/10545552/osd_files.zip)

kirek007, 2023-01-31:
Okey, now I know what's going on. Definitely a bug.

PR #31 (2023-01-31): Fix OSD positioning in PNG files
Fixes #26
closed 2023-01-31 by kirek007