kirek007/ws-osd-py #27: Error on generating images with one of my .osd files

WereWolfPL, 2023-01-21:
Can't generate image files from one of my .osd files (attached in the zip file). It works fine with version 0.5 but every newer release gives error:

```
DEBUG:root:Current: 2800/12570 (fps: 277)
Exception in thread Thread-1 (main):
Traceback (most recent call last):
  File "threading.py", line 1016, in _bootstrap_inner
  File "threading.py", line 953, in run
  File "processor.py", line 632, in main
  File "processor.py", line 536, in __render_osd_frame
  File "processor.py", line 536, in <listcomp>
cv2.error: OpenCV(4.6.0) D:\a\opencv-python\opencv-python\opencv\modules\core\src\matrix_operations.cpp:67: error: (-215:Assertion failed) src[i].dims <= 2 && src[i].rows == src[0].rows && src[i].type() == src[0].type() in function 'cv::hconcat'
```
[AvatarG0004.zip](https://github.com/kirek007/ws-osd-py/files/10473123/AvatarG0004.zip)

kirek007, 2023-01-22:
I guess this is related to #17. Can you try use INav font for inav recording?

WereWolfPL, 2023-01-22:
It's actually betaflight's osd but I've tried also inav fonts and it doesn't work either.
It's strange that it works fine with version 0.5

I compared the images generated with both version and it looks like there was some breakup with the artificial horizon which I had turned on. I attached two frames from my osd: 2878 is the last one that the newer versions could render, 2879 is the next one from version 0.5.
[image]
[image]

kirek007, 2023-01-30:
I've found a bug.. but actually it was a corrupted OSD file. It'll be fixed in next version.

WereWolfPL, 2023-01-30:
> I've found a bug.. but actually it was a corrupted OSD file. It'll be fixed in next version.

Was it because of the artificial horizon? I had problems with two more files, both with AH on. Now I turned it off, but will check it out with the new update.

kirek007, 2023-01-30:
Yup, something definitley around it.

closed 2023-01-31 by kirek007