kirek007/ws-osd-py #28: distance on SRT not updating

iamwinq, 2023-01-22:
I tried rendering the flight from my arduplane setup, you can see here that distance from home is already around 1.5km, but the distance on the SRT overlay is still 0... upon checking the SRT file, I was able to validate that the distance is 0...

[image]

WereWolfPL, 2023-01-22:
Is this with walksnail's high bitrate mode? If yes than it's walksnail's bug, not the overlay tool.

On 25Mbps the distance works fine, but on 50Mbps it shows 0 during the whole flight (at least with my setup).

iamwinq, 2023-01-22:
oh, right, this is on high bitrate... thanks for clarifiying this. should i close this issue i posted?

closed 2023-01-22 by iamwinq