kirek007/ws-osd-py #35: Recognized 'unknown' Software

KrustyFPV, 2023-02-10:
Hi i am using KISS if i drang an OSD file on the overlay tool it tells me this.

do you need some KISS *.osd files for development?

kirek007, 2023-02-10:
Yup, I would need a osd files and the right font. Which flavor of KISS is it?

KrustyFPV, 2023-02-10:
hmm its a Fettec AIO Mini 15A 
This is a short piece of video and the files for it from google 
[AvatarG0088.zip](https://github.com/kirek007/ws-osd-py/files/10710921/AvatarG0088.zip)

i dont`think there is an Font in the Firmware of the FC since it is dedicated OSD just bringing values in to the Walksnail vtx.
if i select auto in googles Font is small if i select fettec font is bigger.

KrustyFPV, 2023-02-10:
If u nee a longer Video where can i Upload?

colzilla, 2023-03-03:
Hi,

I'm running KISS Ultra - and the OSD seems to render larger than the video.
I put some sample files .mp4 .osd .srt and the font /png's at http://colzilla.com/ws-osd/