kirek007/ws-osd-py #36: A CLI tool for batch processing

jksiezni, 2023-02-13:
Hi, could you create a CLI tool for generating OSD? It would greatly improve the workflow with many files.

odgrace, 2023-02-21:
I have made a CLI function to run the process and it seems to work - happy to share/submit PR if there's interest? So far I can make it happen without having to change the main code base which is based around the GUI implementation. One thing I'd ask @kirek007 - at the moment the temporary png files are put into a folder specified by output_path, but also the final output file is a product of that same output_path. Would it not be better to specify only the video out and have the whole temporary png part be hidden in the back?

So the flow would be like:

1. Create temp folder somewhere
2. Make the pngs
3. Make the video with ffmpeg with output file as specified by user
4. Delete the pngs & temp folder

Unless the current implementation already cleans this up and I'm missing it somewhere?

odgrace, 2023-02-24:
Some more ideas for this before I forget:

1. Allowing multiple video path args. So you can process multiple videos in a row then potentially concatenate the outputs
2. FFMPEG config - like CRFs, codecs etc

kirek007, 2023-03-03:
@jksiezni You can checkout latest master and give a try to CLI! :) 
https://github.com/kirek007/ws-osd-py#cli

jksiezni, 2023-03-03:
Ok, the CLI tool works great on my Ubuntu 22.04, but I have noticed some issues after a few quick runs.

1. The `--include-srt` fails when running the tool from a working dir that does not contain _font.ttf_
```
$ python3 ~/git/hub/ws-osd-py/cli.py --video-path AvatarG0019.mp4 --font-path ~/git/hub/ws-osd-py/sneaky/WS_BF_24.png --remove-png --include-srt
Traceback (most recent call last):
  File "/home/jakub/git/hub/ws-osd-py/cli.py", line 100, in <module>
    gen.main()
  File "/home2/jakub/git/hub/ws-osd-py/processor.py", line 678, in main
    frame_osd_srt = Utils.overlay_srt_line(self.config.fast_srt, osd_frame_no_srt, srt_data["line"], self.font.get_srt_font_size(
  File "/home2/jakub/git/hub/ws-osd-py/processor.py", line 340, in overlay_srt_line
    return Utils.overlay_srt_line_slow(img, line, font_size, left_offset)
  File "/home2/jakub/git/hub/ws-osd-py/processor.py", line 347, in overlay_srt_line_slow
    font = ImageFont.truetype("font.ttf", font_size)
  File "/home2/jakub/git/hub/ws-osd-py/.venv/lib/python3.10/site-packages/PIL/ImageFont.py", line 976, in truetype
    return freetype(font)
  File "/home2/jakub/git/hub/ws-osd-py/.venv/lib/python3.10/site-packages/PIL/ImageFont.py", line 973, in freetype
    return FreeTypeFont(font, size, index, encoding, layout_engine)
  File "/home2/jakub/git/hub/ws-osd-py/.venv/lib/python3.10/site-packages/PIL/ImageFont.py", line 249, in __init__
    self.font = core.getfont(
OSError: cannot open resource

```
  Should the font be somehow packaged/installed beforehand? If it worked so far for GUI, so maybe add another param, or add an environment variable, ie. WS_FONT_TTF.

2. The `--hide-sensitive-osd` option didn't hide relevant data, but instead it hid throttle pos and a blackbox status. :smile: Maybe, I should create a new issue for that.

@odgrace If you want, I can make a shell script that takes output from the tool and concatenates videos. In that case it would be very useful to have control over ffmpeg args passed by the CLI tool, so it could output raw video to avoid recompressing.

Thanks for making this tool!
Świetna robota. ;)

kirek007, 2023-03-03:
- Indeed font.tff is part of release so you can take it from release package.
- hide sensitive data works only with inav
:)

odgrace, 2023-03-04:
> 1. The `--include-srt` fails when running the tool from a working dir that does not contain _font.ttf_
 
This is laziness on my part - pretty easy fix. For now just use the full path to the font

> @odgrace If you want, I can make a shell script that takes output from the tool and concatenates videos. In that case it would be very useful to have control over ffmpeg args passed by the CLI tool, so it could output raw video to avoid recompressing.

There are two parts to this. 

1. The concat should be pretty easy and can be done after rendering each without re-encoding. That part can happen through modifying ```cli.py```, I just need to change the video/osd/srt args to accept multiple files or like a shell expansion.
2. FFMPEG settings are a bit more involved and will have to delve into the ```processor.py``` code to get OsdGenerator.main to accept arguments. Rather than exhaustively re-doing all the ffmpeg args in the cli args it would be nice to somehow maybe passthrough? There would need to be some thinking to allow nearly everything ffmpeg does since the position of arguments matters a lot. Easier option: be willing to accept a limited set like encoder (h264/h265) and a crf then leave it at that? At the moment it's ```-crf 0``` which is overkill and ```-b:v 40M``` which arguably has no effect given the CRF. Maybe add some ```-vf``` as a passthrough if someone has a genuine useless

odgrace, 2023-03-04:
https://github.com/kirek007/ws-osd-py/pull/44

I was wrong - the srt is not found because the GUI packages the file up and moves to a root folder. This PR finds a fallback location on OSError. I've tried this from another folder and it seems to work.