kirek007/ws-osd-py #37: Interface suggestion: Prompt for font file automatically after loading in video, OSD and SRT file

Painless360, 2023-02-15:
Hi! 

Love the tool! I was wondering if it would be possible to add a prompt for when the app started to guide new users through the process of importing the key files. In particular the font file. Or, have the app select a default font based on the video file format so that the OSD image appears as soon as the video file is imported? 

Thanks for reading this, I am a fan of this project and will make a video soon! 

Lee
P.S. I've tried to send you some money using the 'coffee' app but it will not work, happy to PayPal you some cash instead if you contact me at lee@painless360.com :D

kirek007, 2023-02-16:
Hey, yeah the UX is not the best :D I was not expecting that big audience. That is a good suggestion, but I'm planning to include default fonts from the list (plus selecting the right one based on FC data).
There is no ETA for the next release, but UX improvements are high on my list.

PS. Don't bother yourself with Paypal, Thank you is as good as sending virtual coffee ;)  
Cheers, Kirek.

Painless360, 2023-02-18:
Hey Kirek! The video on the channel today should help raise the awareness of this excellent project a little more. Happy to support you with a donation, this is excellent work! Are you replying to comments on the video? The channel name wasn't the same as on here so wanted to make sure it was you. Happy flying!

kirek007, 2023-02-19:
@Painless360, MindTheGapFPV is my youtube channel for fpv :)
I can already see a increased issue tickets 😂 Thanks!