kirek007/ws-osd-py #39: Add ability to 'widescreen' the OSD elements in the X axis for Walksnail so OSD elements are at side of screen

Painless360, 2023-02-18:
Hello again! Love the tool and thank you so much for sharing it with the community! 

One thing I like with DJI is that the OSD elements can be pushed to the very sides and top of the screen and that results in a less cluttered video. With Walksnail the OSD elements sit in a 4:3 box that you can move and resize using the tools in the app (thank you again).

One thing that I'd welcome is the ability to stretch the elements on the X axis to move them into the sides of the screen to leave the centre less cluttered. By using a multiplier applied for OSD elements from the centre of the normal layout they could be pushed into a more 16:9 format from that default 4:3. 

I hope that makes sense.

kirek007, 2023-02-19:
I really hope that this won't be a problem once everyone will start using BF4.4 that supports HD layout, so if this will be still problem in some time I'll take a look into that.

ShikOfTheRa, 2023-03-06:
It works great in BF4.4 rc version. No offsets / zoom 

https://www.youtube.com/watch?v=sR2HKXZ9r9w

Painless360, 2023-03-07:
That is great to see! Thanks for posting. I fly INAV here and that doesn't work the same way so this feature would be nice to have.

odgrace, 2023-03-08:
@Painless360 I'm just wondering - I run INAV 6 and my OSD elements work in 16:9.

[image]

odgrace, 2023-03-09:
Hi Lee,

As I recall - in the ports tab the UART is setup as MSP DisplayPort. Under OSD there is an option for Avatar that correctly sizes the OSD layout. Then the last step I think is in the goggles to move the OSD position to accommodate.

Looking at the INAV source I think that came in around 6.0 FP1 or RC1. 

Cheers,
Owen

Painless360, 2023-03-09:
Hey Owen, 

Thanks, that is really interesting. Adding the X axis stretch feature would support all those pilots not using the latest BF, INAV or something like Ardupilot and allow pilots to tweak the 'widescreen' look of the OSD when applied.

Thanks for considering and coming back to me. 

Happy flying!

Lee

odgrace, 2023-03-09:
So the thing about the OSD elements is that the they are just glyphs that are rendered on a 2d array. This program and our goggles don't know _what_ the elements are - just they are told to render glyph 'A' at x,y. As you say you could increase the spacing between the glyphs so that

```179mah```

becomes

```1  7  9  mah```

That would be the 'easiest' implementation but I don't think that would solve the intent (clear up OSD). Another option is to take the first n and last n columns and move them to the extremes - but there's a risk that long elements get chopped in the process.

The final one which requires the most effort but has the best results is to somehow 'grab' elements from the left and the right and move them to the respective sides. That's fairly complex and knowing that in a few weeks INAV, BF and ARDU are all about to release versions which support this, I get a feeling adding the feature would be temporary only!

Notwithstanding, do you have a 4:3 mp4/osd which I can have a look at? Preferably a small one...

kirek007, 2023-03-09:
I think that @Painless360 would like to see a FakeHD like it is implemented in https://github.com/fpv-wtf/msp-osd/#fakehd. I've mixed feelings about it.

avsaase, 2023-03-09:
> I think that @Painless360 would like to see a FakeHD like it is implemented in https://github.com/fpv-wtf/msp-osd/#fakehd. I've mixed feelings about it.

How could that ever work with the reasons @odgrace described? You can't chop up the grid without potentially breaking up OSD elements.

kirek007, 2023-03-10:
This assumes that while you're placing OSD elements you're aware of splits.
So in perfect condition none of the elements would be splited.