kirek007/ws-osd-py #40: macOS compile error

Radla, 2023-02-18:
Now it works with my python version, but i get an error, short after downloading wxPython:
(macOS 13.1, M1)

`makefile:1: Makefile.venv: No such file or directory
curl \
		-o Makefile.fetched \
		-L "https://github.com/sio/Makefile.venv/raw/v2022.07.20/Makefile.venv"
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100  7159  100  7159    0     0   2300      0  0:00:03  0:00:03 --:--:--  4370
echo "147b164f0cbbbe4a2740dcca6c9adb6e9d8d15b895be3998697aa6a821a277d8 *Makefile.fetched" \
		&& mv Makefile.fetched Makefile.venv
147b164f0cbbbe4a2740dcca6c9adb6e9d8d15b895be3998697aa6a821a277d8 *Makefile.fetched
python3 -m venv ./.venv
./.venv/bin/python -m pip install --upgrade pip setuptools wheel
Requirement already satisfied: pip in ./.venv/lib/python3.11/site-packages (22.3.1)
Collecting pip
  Downloading pip-23.0.1-py3-none-any.whl (2.1 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 2.1/2.1 MB 6.5 MB/s eta 0:00:00
Requirement already satisfied: setuptools in ./.venv/lib/python3.11/site-packages (65.5.0)
Collecting setuptools
  Downloading setuptools-67.3.2-py3-none-any.whl (1.1 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.1/1.1 MB 6.3 MB/s eta 0:00:00
Collecting wheel
  Using cached wheel-0.38.4-py3-none-any.whl (36 kB)
Installing collected packages: wheel, setuptools, pip
  Attempting uninstall: setuptools
    Found existing installation: setuptools 65.5.0
    Uninstalling setuptools-65.5.0:
      Successfully uninstalled setuptools-65.5.0
  Attempting uninstall: pip
    Found existing installation: pip 22.3.1
    Uninstalling pip-22.3.1:
      Successfully uninstalled pip-22.3.1
Successfully installed pip-23.0.1 setuptools-67.3.2 wheel-0.38.4
./.venv/bin/pip install -r requirements.txt
Collecting altgraph==0.17.3
  Using cached altgraph-0.17.3-py2.py3-none-any.whl (21 kB)
Collecting ffmpeg-python==0.2.0
  Using cached ffmpeg_python-0.2.0-py3-none-any.whl (25 kB)
Collecting future==0.18.2
  Using cached future-0.18.2.tar.gz (829 kB)
  Preparing metadata (setup.py) ... done
Collecting numpy==1.24.1
  Downloading numpy-1.24.1-cp311-cp311-macosx_11_0_arm64.whl (13.8 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 13.8/13.8 MB 7.0 MB/s eta 0:00:00
Collecting opencv-python==4.6.0.66
  Using cached opencv_python-4.6.0.66-cp37-abi3-macosx_11_0_arm64.whl (30.0 MB)
Collecting pefile==2022.5.30
  Using cached pefile-2022.5.30.tar.gz (72 kB)
  Preparing metadata (setup.py) ... done
Collecting Pillow==9.3.0
  Downloading Pillow-9.3.0-cp311-cp311-macosx_11_0_arm64.whl (2.9 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 2.9/2.9 MB 6.9 MB/s eta 0:00:00
Collecting pyinstaller==5.7.0
  Using cached pyinstaller-5.7.0-py3-none-macosx_10_13_universal2.whl (923 kB)
Collecting pyinstaller-hooks-contrib==2022.14
  Using cached pyinstaller_hooks_contrib-2022.14-py2.py3-none-any.whl (252 kB)
Collecting Pypubsub==4.0.3
  Using cached Pypubsub-4.0.3-py3-none-any.whl (61 kB)
Collecting pywin32-ctypes==0.2.0
  Using cached pywin32_ctypes-0.2.0-py2.py3-none-any.whl (28 kB)
Collecting six==1.16.0
  Using cached six-1.16.0-py2.py3-none-any.whl (11 kB)
Collecting wxPython==4.2.0
  Downloading wxPython-4.2.0.tar.gz (71.0 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 71.0/71.0 MB 6.8 MB/s eta 0:00:00
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [8 lines of output]
      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "/private/var/folders/nx/f2jsnvsj0p1_420lxfncrq4r0000gn/T/pip-install-3f8fsw94/wxpython_3e1edf12fc2044d6b8cb6c5165d99e83/setup.py", line 27, in <module>
          from buildtools.config import Config, msg, opj, runcmd, canGetSOName, getSOName
        File "/private/var/folders/nx/f2jsnvsj0p1_420lxfncrq4r0000gn/T/pip-install-3f8fsw94/wxpython_3e1edf12fc2044d6b8cb6c5165d99e83/buildtools/config.py", line 30, in <module>
          from attrdict import AttrDict
      ModuleNotFoundError: No module named 'attrdict'
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
make: *** [.venv/bin/.initialized-with-Makefile.venv] Error 1
`

Radla, 2023-02-18:
After some research i found some infos about problems with Python 3.11 and wxPython. So i changed to Python 3.10.4 and try to install attrict , but i have still no success:
(btw. i dont know, why the code  tag does not work, sorry!)
`
ralle@Mac-mini ws-osd-py % pip3 install attrdict3                                                         
Requirement already satisfied: attrdict3 in /Users/ralle/.pyenv/versions/3.10.4/lib/python3.10/site-packages (2.0.2)
Requirement already satisfied: six in /Users/ralle/.pyenv/versions/3.10.4/lib/python3.10/site-packages (from attrdict3) (1.16.0)
ralle@Mac-mini ws-osd-py % pip3 install attrdict 
Collecting attrdict
  Using cached attrdict-2.0.1-py2.py3-none-any.whl (9.9 kB)
Requirement already satisfied: six in /Users/ralle/.pyenv/versions/3.10.4/lib/python3.10/site-packages (from attrdict) (1.16.0)
Installing collected packages: attrdict
Successfully installed attrdict-2.0.1
ralle@Mac-mini ws-osd-py % pip install attrdict 
Requirement already satisfied: attrdict in /Users/ralle/.pyenv/versions/3.10.4/lib/python3.10/site-packages (2.0.1)
Requirement already satisfied: six in /Users/ralle/.pyenv/versions/3.10.4/lib/python3.10/site-packages (from attrdict) (1.16.0)
ralle@Mac-mini ws-osd-py % make run                                                                       
./.venv/bin/pip install -r requirements.txt
Collecting altgraph==0.17.3
  Using cached altgraph-0.17.3-py2.py3-none-any.whl (21 kB)
Collecting ffmpeg-python==0.2.0
  Using cached ffmpeg_python-0.2.0-py3-none-any.whl (25 kB)
Collecting future==0.18.2
  Using cached future-0.18.2.tar.gz (829 kB)
  Preparing metadata (setup.py) ... done
Collecting numpy==1.24.1
  Using cached numpy-1.24.1-cp311-cp311-macosx_11_0_arm64.whl (13.8 MB)
Collecting opencv-python==4.6.0.66
  Using cached opencv_python-4.6.0.66-cp37-abi3-macosx_11_0_arm64.whl (30.0 MB)
Collecting pefile==2022.5.30
  Using cached pefile-2022.5.30.tar.gz (72 kB)
  Preparing metadata (setup.py) ... done
Collecting Pillow==9.3.0
  Using cached Pillow-9.3.0-cp311-cp311-macosx_11_0_arm64.whl (2.9 MB)
Collecting pyinstaller==5.7.0
  Using cached pyinstaller-5.7.0-py3-none-macosx_10_13_universal2.whl (923 kB)
Collecting pyinstaller-hooks-contrib==2022.14
  Using cached pyinstaller_hooks_contrib-2022.14-py2.py3-none-any.whl (252 kB)
Collecting Pypubsub==4.0.3
  Using cached Pypubsub-4.0.3-py3-none-any.whl (61 kB)
Collecting pywin32-ctypes==0.2.0
  Using cached pywin32_ctypes-0.2.0-py2.py3-none-any.whl (28 kB)
Collecting six==1.16.0
  Using cached six-1.16.0-py2.py3-none-any.whl (11 kB)
Collecting wxPython==4.2.0
  Using cached wxPython-4.2.0.tar.gz (71.0 MB)
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [8 lines of output]
      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "/private/var/folders/nx/f2jsnvsj0p1_420lxfncrq4r0000gn/T/pip-install-ku89sni3/wxpython_1f8508a9b08242399b851761842bf16e/setup.py", line 27, in <module>
          from buildtools.config import Config, msg, opj, runcmd, canGetSOName, getSOName
        File "/private/var/folders/nx/f2jsnvsj0p1_420lxfncrq4r0000gn/T/pip-install-ku89sni3/wxpython_1f8508a9b08242399b851761842bf16e/buildtools/config.py", line 30, in <module>
          from attrdict import AttrDict
      ModuleNotFoundError: No module named 'attrdict'
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
make: *** [.venv/bin/.initialized-with-Makefile.venv] Error 1
`

kirek007, 2023-02-18:
> File "", line 2, in
> File "", line 34, in

You have to download pre-built package and install it before make run
You can find a package here https://pypi.org/project/wxPython/#files

kirek007, 2023-02-18:
Try follow this
https://github.com/kirek007/ws-osd-py#common-issues-for-linux

Radla, 2023-02-18:
Thank you, i downloaded the package but i still get the same error:

`ralle@Mac-mini ws-osd-py % source .venv/bin/activate
(.venv) ralle@Mac-mini ws-osd-py % pip install -f ./wxPython-4.2.0-cp310-cp310-macosx_10_10_universal2.whl wxPython
Looking in links: ./wxPython-4.2.0-cp310-cp310-macosx_10_10_universal2.whl
Collecting wxPython
  Using cached wxPython-4.2.0.tar.gz (71.0 MB)
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [8 lines of output]
      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "/private/var/folders/nx/f2jsnvsj0p1_420lxfncrq4r0000gn/T/pip-install-18eflkd8/wxpython_c6fa2fdd99fa4e679605cbf613973ac0/setup.py", line 27, in <module>
          from buildtools.config import Config, msg, opj, runcmd, canGetSOName, getSOName
        File "/private/var/folders/nx/f2jsnvsj0p1_420lxfncrq4r0000gn/T/pip-install-18eflkd8/wxpython_c6fa2fdd99fa4e679605cbf613973ac0/buildtools/config.py", line 30, in <module>
          from attrdict import AttrDict
      ModuleNotFoundError: No module named 'attrdict'
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
`

kirek007, 2023-02-18:
> (.venv) ***@***.*** ws-osd-py % pip install -f

Please try to remove .venv directory and try from zero.. If not I don't
know.. Sorry :(

Radla, 2023-02-18:
Maybee it has something to do with my Apple Silicon M1 processor? There are only 'universal' binaries, nothing named 'arm'.

Radla, 2023-02-18:
i removed the .venv folder und now it seems to work!
(Only a gui problem, which i should solve)

`./.venv/bin/python -c 'import sys; valid=(sys.version_info > (3,9) and sys.version_info < (3,11)); sys.exit(0) if valid else sys.exit(1)' || (echo "Python 3.10 is required"; exit 1)
./.venv/bin/python osd_gui.py
This program needs access to the screen. Please run with a
Framework build of python, and only when you are logged in
on the main display of your Mac.
make: *** [run] Error 1
(.venv) ralle@Mac-mini ws-osd-py % `

closed 2023-02-19 by kirek007

Radla, 2023-02-19:
O.K. after some hours of experimention with this GUI issue, i finally give up :(

kirek007, 2023-02-19:
Sorry to hear, but I dont have any macos to help you investigate the root cause. I saw multiple users using app in macos.