kirek007/ws-osd-py #41: Feature Request: Render OSD as video ONLY (transparent background)

Pairan, 2023-02-19:
Currently I'm struggling a bit with the intended workflow (v0.9). My DVR videos are about 9mins long and as such create a humongous pile of PNG files as I choose to render PNG only.

These 9min of DVR footage result in about 35840 PNG files ... to process these within the video editor of choice is not much of a pleasant task :) So here is my suggestion

### add an option where the OSD is generated as a video with transparent background ONLY
That would make file handling MUCH easier and not even cost to much changes in the code .. e.g.

if option `transparent osd video` is checked then return a transparent frame instead of the frame in the video.

```Python
    def read_frame(self):
        ret, frame = self.videoFile.read()
        if not ret:
            return None

        if len(frame) == 0:
            return None

       """
            could look like this perhaps:
       """
       if transparent_osd_video:
            return transparentFrame

        return VideoFrame(frame)
```

### Benefits
- The resulting video would be easy to overlay over ANY format and content 
- It's only ONE file against several 35840 per 9min (I fly long trips, ... yes!)

kirek007, 2023-02-19:
Hey, I think that you're importing png sequences in the wrong way :) usually (premiere pro, DaVinci and vega) are able to import all png in a few seconds as a clip. Which video editor are you using? 

But, you're right, PNGs are heavy in terms of used space. I'll take that into account :)

Pairan, 2023-02-19:
Using Final Cut Pro X in the latest version on a Mac Studio with 32GB RAM and the not so slow M1 MAX :)

kirek007, 2023-02-19:
I have no experience with this one :)

Pairan, 2023-02-19:
Additional observation on this: the last 40 PNG files seem to to be corrupt! ... if I skip those out, the data is processed.

FCPX (Final Cut Pro X) is capable of doing such imports and the handling is exact as described in some other products: Drag all files in, add them to the time line and align the combined file with the (dvr) video.

kirek007, 2023-03-14:
@Pairan Hey, just a heads up that I'm working on it, and looks like I'm able to do so. So PNG files will be gone very soon :)

Pairan, 2023-03-15:
Sounds good to me :) Let me know, when I can give it a try