kirek007/ws-osd-py #42: Can't run app on macOS

nogod4me, 2023-02-19:
Can't rum app on my MacBook M1 (macOS Ventura 13.1), I've tried using the instructions

```
ivanprivalihin@MacBook-Pro-Ivan ~ % cd ws-osd-py
ivanprivalihin@MacBook-Pro-Ivan ws-osd-py % make run    
Python 3.9.6 (default, Oct 18 2022, 12:41:40) [Clang 14.0.0 (clang-1400.0.29.202)]
pip 23.0.1 from /Users/ivanprivalihin/ws-osd-py/.venv/lib/python3.9/site-packages/pip (python 3.9)
venv: ./.venv
./.venv/bin/python -c 'import sys; valid=(sys.version_info > (3,9) and sys.version_info < (3,11)); sys.exit(0) if valid else sys.exit(1)' || (echo "Python 3.10 is required"; exit 1)
./.venv/bin/python osd_gui.py
Traceback (most recent call last):
  File "/Users/ivanprivalihin/ws-osd-py/osd_gui.py", line 4, in <module>
    from processor import OSDFile, OsdFont, OsdPreview, VideoFile
  File "/Users/ivanprivalihin/ws-osd-py/processor.py", line 108
    match self.fcType:
          ^
SyntaxError: invalid syntax
make: *** [run] Error 1
```

I'm not a programmer so it will be cool have a compiled self-installing package for it. It's a good manners rule to have all platforms assets. Thanks.

kirek007, 2023-02-19:
It looks like you are using python older than 3.10. Can you try instal
exactly 3.10?

kirek007, 2023-02-19:
As it is stated in a log file, you're using "Python 3.9.6" which is not supported. Please take a look into "common issues with linux" in Readme.

closed 2023-02-19 by kirek007

nogod4me, 2023-02-19:
Already tried 3.10 and 3.11 but it uses older 3.9, why? Please, give exact instructions to fix it, I don’t understand python and terminal tricks(

```
ivanprivalihin@MacBook-Pro-Ivan ~ % python3 --version
Python 3.10.5
ivanprivalihin@MacBook-Pro-Ivan ~ % cd ws-osd-py     
ivanprivalihin@MacBook-Pro-Ivan ws-osd-py % make run         
Python 3.9.6 (default, Oct 18 2022, 12:41:40) [Clang 14.0.0 (clang-1400.0.29.202)]
pip 23.0.1 from /Users/ivanprivalihin/ws-osd-py/.venv/lib/python3.9/site-packages/pip (python 3.9)
venv: ./.venv
./.venv/bin/python -c 'import sys; valid=(sys.version_info > (3,9) and sys.version_info < (3,11)); sys.exit(0) if valid else sys.exit(1)' || (echo "Python 3.10 is required"; exit 1)
./.venv/bin/python osd_gui.py
Traceback (most recent call last):
  File "/Users/ivanprivalihin/ws-osd-py/osd_gui.py", line 4, in <module>
    from processor import OSDFile, OsdFont, OsdPreview, VideoFile
  File "/Users/ivanprivalihin/ws-osd-py/processor.py", line 108
    match self.fcType:
          ^
SyntaxError: invalid syntax
make: *** [run] Error 1
```