kirek007/ws-osd-py #53: Error message after .png generation

fpvandi, 2023-03-27:
Hello together,

when i try to render my videos from the internal hard drive i get an error message that there are no .png available. When I then go into the AvatarGxxxx_generated folder it is actually empty. As soon as the video files and the .srt and .osd files are on another drive (e.g. external SSD or microSd card) I can render the videos normally.
Is this error known or is it just me?

[image]

kirek007, 2023-03-27:
Hey, I'm quite sure that this is relate to space in directory path. Sadly
this is bug in library I'm using and can't be fixed today :(
Please try to generate in different directory

kirek007, 2023-03-27:
Duplicate of #23

closed 2023-03-27 by kirek007