kirek007/ws-osd-py #55: Typo in displaying the SRT data: "disatnce" vs "distance"

SplinterII, 2023-04-08:
In the rendered video the Walksnail own OSD data has for the distance indicator the label "disatnce", instead of "distance".

kirek007, 2023-04-08:
Hey, please use latest latest version :)
You can find it here:
https://github.com/kirek007/ws-osd-py/releases/tag/v0.9.0-beta

SplinterII, 2023-04-08:
[image]

In the latest version I still see it?

kirek007, 2023-04-08:
I'm 100prc sure that you're not using latest version form provided link :)
please double check and download latest version from link I've provided
above

SplinterII, 2023-04-08:
I was looking at the initial/first render, the render with the new version is ok indeed! Deleted my screenshot when I found out….

 

First time user, sorry for the inconvenience 😉

 

Happy Easter!